lucullusTheOnly/TinyWire #1: Does this library support slave clock stretching?

stuartpittaway, 2017-11-20:
When running on an ATTINY85 does the slave implement clock stretching correctly?

lucullusTheOnly, 2017-11-23:
I'm not completely sure about that, but I found corresponding parts in the datasheet of the ATTiny85:
When the USI hardware detects a start condition, it holds the clock low, until the flag is resetted. The same is happening, when a counter overflow occurs after 1 byte was transmitted. The library is resetting the bits near the end of the corresponding ISR. So this should work.
The datasheet also says, that "Two-wire Clock Control Unit will hold the SCL low until the
slave is ready to receive more data." (in 15.3.6 Clock speed considerations). So it seems, that the USI hardware itself uses clock streching for receiving every bit, if necessary.

All in all I would say: Yes, it does.

And additionally the Master Code also waits for the clock to go high. (done by a while loop in USI_TWI_Master_Transfer() and the if conditions in USI_TWI_Master_Start/Stop(), when SIGNAL_VERIFY is defined)

Please correct me, if I'm wrong

lucullusTheOnly, 2018-01-30:
I'll close this issue for now, because there has not been any activity since my post.

closed 2018-01-30 by lucullusTheOnly