lucullusTheOnly/TinyWire #13: Wont work in atmel studio

rharrison, 2018-09-01:
So I have this working fine in arduino ide but when i move to Atmel studio and compile it wont work even though it compiles and uploads with no errors. This is using an atTimey85 with F_CPU=1000000L in Atmel Studio. Any thoughts from those cleverer than me?

TIA 

Robert

lucullusTheOnly, 2019-03-24:
How does it not work? Can you check, what actually happens by adding a live signal and maybe some indicators to check, where exactly the problem happens? That would make it drastically easier to find the problem.

Did it work with a clock speed of 1MHz when using the Arduino IDE?

lucullusTheOnly, 2020-12-21:
I'm closing this issue, as it's unlikely to get further information 2 years after the opening of the issue

closed 2020-12-21 by lucullusTheOnly