lucullusTheOnly/TinyWire #22: Problem ... ESP8266 -> ATTiny85 ...

Humancell, 2019-07-21:
Hello,

I'm hitting an issue that is driving me nuts.  :-)

1. I have an ATTiny85 that is running this code:  [Qwiic_Loudness_Sensor_Firmware.ino](https://github.com/ZIOCC/Zio-Qwiic-Loudness-Sensor-I2C/blob/master/Codes/Qwiic_Loudness_Sensor_Firmware/Qwiic_Loudness_Sensor_Firmware/Qwiic_Loudness_Sensor_Firmware.ino)

2. I have an Arduino UNO (and MEGA) running this code:  [Example1-Basic_Reading/Example1-Basic_Reading.ino](https://github.com/ZIOCC/Zio-Qwiic-Loudness-Sensor-I2C/blob/master/Codes/Qwiic%20Loudness%20Sensor%20Examples/Example1-Basic_Reading/Example1-Basic_Reading.ino)

This works perfectly.

3. When I put this same code on an ESP8266 it fails: [Example1-Basic_Reading/Example1-Basic_Reading.ino](https://github.com/ZIOCC/Zio-Qwiic-Loudness-Sensor-I2C/blob/master/Codes/Qwiic%20Loudness%20Sensor%20Examples/Example1-Basic_Reading/Example1-Basic_Reading.ino)

My ESP8266 can talk with countless I2C devices, and even an Arduino UNO running some Wire slave code.  It just can't read from the ATTiny85 running that code above.

Specifically ... it's the requestFrom() operation that fails every time ... always returning 0 bytes.  If I comment out the get_value(); call that is in loop() then the LED blinks just fine.

Upon searching (for hours) I found some people referencing this post from the /rambo library that you mention:  https://github.com/rambo/TinyWire/issues/36

I'm wondering if there is a similar fix that you have made, or might have to make, in your library?

Any ideas on debugging or what might be going on?

Humancell, 2019-08-05:
Hello?  Any ideas on how I can best debug this scenario?

tatobari, 2019-08-05:
Are you using pull-up resistors? What's their value?

Humancell, 2019-08-05:
Yes ... I've got pull-ups in place, and they are 4.7k ohm.

tatobari, 2019-08-05:
Sorry for the dumb questions about stuff you've probably checked a million times already. I'll try using a Tiny85 with an ESP8266, but, before, I gotta ask. Again, sorry:

1. Are your wires on the correct pins? PB0 is SDA and PB2 is SDA, got that one right? (Sorry sorry sorry sorry for asking this)
1. Have you been able to use the ATTiny85's I2C as a master with some other device? (Another dumb question, sorry)
1. Looks like you're using Arduino IDE and not AtmelStudio. What ATTiny85 library are you using? Is it ATTinyCore?

Humancell, 2019-08-05:
Hello,

There are no dumb questions ... you're helping to debug this so everything is fair game.  :-)

The board (running the I2C Slave code) is from here:  https://www.smart-prototyping.com/Zio-Qwiic-Loudness-Sensor-I2C?search=sound

Here is the schematic of the board:  https://github.com/ZIOCC/Zio-Qwiic-Loudness-Sensor-I2C/blob/master/zio%20qwiic%20loudness%20sensor%20schematic.pdf

And the firmware for the ATTiny on the board is here:  https://github.com/ZIOCC/Zio-Qwiic-Loudness-Sensor-I2C/blob/master/Codes/Qwiic_Loudness_Sensor_Firmware/Qwiic_Loudness_Sensor_Firmware/Qwiic_Loudness_Sensor_Firmware.ino

I've cabled it according to this page:  https://www.smart-prototyping.com/Qwiic.html

It *appears* to be built correctly, and configured correctly.

1. The wiring appears to be correct.  When running the example code that I referenced above, the LED is being controlled correctly ... so the "writes" appear to work:

```
void ledOn() {
  Wire.beginTransmission(qwiicAddress);
  Wire.write(COMMAND_LED_ON);
  Wire.endTransmission();
}

void ledOff() {
  Wire.beginTransmission(qwiicAddress);
  Wire.write(COMMAND_LED_OFF);
  Wire.endTransmission();
}
```

And I am using the Arduino IDE, however I have not yet recompiled the code nor uploaded to the board until I determined if there were other hardware issues going on.  Which board library (url?) would you recommended be added to the current IDE?

Humancell, 2019-08-05:
Sorry ... and again, I used this board with an UNO, Mega, and Pro Mini without an issue ... the exact same off-the-shelf board running the exact same ATTiny85 code ... and it seems to work perfectly.  It's only when I connect it to the ESP-12F module that I don't get a response.  And my ESP-12F module can talk with all sorts of other I2C devices ...

tatobari, 2019-08-07:
Last one, which ESP8266 board are you using?

Humancell, 2019-08-07:
I'm actually using a very simple prototyping board from ESProto ... http://www.esproto.com/product/esproto-one-kit/

It's really just a ESP-12F on a PCB ... very raw.

tatobari, 2019-08-07:
So:
- ATTiny85 works perfectly well as a slave using Arduino UNO, MEGA and Pro Mini as Masters.
- ESP-12F Breakout Board works perfectly well as Master with countless I2C slave devices.
- However, they don't work correctly together. All you get is ones.

Right?

Not an easy one. I keep thinking about issues related to an electric issue. I'll test this during the weekend (I'm out of town now) and get back to you.

Humancell, 2019-08-07:
Yes ... with one more bullet:

- I2C *writes* (ESP-12F -> ATTiny85) appear to work since the I2C commands COMMAND_LED_ON, and COMMAND_LED_OFF are working!  Which blows my mind.

It's only when I go to read that the reads fail.

I do know this is not an easy one ... and truly appreciate your time on this.  I'm going to have to find your donate button if we can solve this.  Let me know if there are other things I can provide.

tatobari, 2019-08-07:
Haha, if you knew how far way from setting up a donate button I am. I'll get back to you during the weekend. This kind of issues are always fun.

tatobari, 2019-08-07:
You've probably already checked this out.

https://arduino-esp8266.readthedocs.io/en/2.5.2/boards.html#generic-esp8266-module

Humancell, 2019-09-20:
Hello,

I wanted to check back and see if you ever got a chance to experiment?

I did review that link that you provided ... was there something in particular that you wanted me to read?

Just as a side note, we PCB that we designed around the ESP-12F module, and have ~15 different sensor products that I've designed and built for use in a wide range of applications.  I have written a lot of code for the ESP-12F, interacting with numerous I2C devices.  This is the first time I've run into an issue like this, which is why I was looking for some tips/input on what might not be working.

I'm hoping that you had a chance to investigate a little ... or maybe have some ideas on requirements to be able to read properly from a ATTiny85 from the ESP-12F.  I spent some time Googling for ideas, and experimented with the code some ... but can still *write* ... just can't *read* ...  :-?

Humancell, 2019-12-12:
Hello!

I wanted to let you know that I just resolved this issue!  :-)

It turns out that I needed to increase the "clock stretching" on the ESP8266 I2C bus, adding the following line to my code:

> Wire.setClockStretchLimit(1000); // Allow for 1000us of clock stretching

The default Arduino ESP8266 StretchLimit is 230us ... which I guess was just too short for the board running your code.  I increased this to 1000us and it's working great!

Thanks!

closed 2019-12-12 by Humancell