lucullusTheOnly/TinyWire #28: ATTINY85 not working wirh both "TinyWire.h" and "DigiKeyboard"

Giova02, 2021-09-14:
Hi to everyone!
That's my first post here.
I'm trying to write presaved messages controling a digispark attiny85 using i2c communication with an esp01 in AP mode to control the system via smartphone.
Doing some tests I can now do everything I wanted to do but now, after adding the DigiKeyboard commands and lib, when I tried uploading the following I got this error..

`C:\Users\10293_~1\AppData\Local\Temp\arduino_build_994554\libraries\TinyWire-master\twi.cpp.o: In function `__vector_2':
C:\Users\10293_000\Documents\Arduino\libraries\TinyWire-master/twi.cpp:822: multiple definition of `__vector_2'
C:\Users\10293_~1\AppData\Local\Temp\arduino_build_994554\libraries\DigisparkKeyboard\usbdrvasm.S.o:C:\Users\10293_000\Documents\ArduinoData\packages\digistump\hardware\avr\1.6.7\libraries\DigisparkKeyboard/usbdrvasm165.inc:41: first defined here
collect2.exe: error: ld returned 1 exit status
Using library DigisparkKeyboard in folder: C:\Users\10293_000\Documents\ArduinoData\packages\digistump\hardware\avr\1.6.7\libraries\DigisparkKeyboard (legacy)
Using library TinyWire-master in folder: C:\Users\10293_000\Documents\Arduino\libraries\TinyWire-master (legacy)
exit status 1
Error compiling for board Digispark (Default - 16.5mhz).
`

Can someone help me to solve this pls?

Here's also the attiny85 code (I don't think it's a problem related to the esp01 code so i think that the attiny85 one could be enough)

`#include "KeyboardIT.h"
#include <TinyWire.h>
#define own_address 9

void setup()
{
 TinyWire.begin(own_address);
 TinyWire.onReceive(comando);
  pinMode(0, OUTPUT); //LED on Model B
  pinMode(1, OUTPUT); //LED on Model A  or Pro
}

void loop()
{
}

void comando(int dato)
{
 while (TinyWire.available()) 
 {
  char c = TinyWire.read();
  if(c=='1')
  {
  digitalWrite(0, HIGH);
  digitalWrite(1, HIGH);
  }
  else if(c=='2')
  {
  DigiKeyboard.sendKeyStroke(0);
  DigiKeyboard.delay(2000);
  DigiKeyboard.sendKeyStroke(KEY_H);
  DigiKeyboard.delay(2000);
  DigiKeyboard.sendKeyStroke(KEY_E);
  DigiKeyboard.delay(2000);
  DigiKeyboard.sendKeyStroke(KEY_L);
  DigiKeyboard.delay(2000);
  DigiKeyboard.sendKeyStroke(KEY_L);
  DigiKeyboard.delay(2000);
  DigiKeyboard.sendKeyStroke(KEY_O);
  DigiKeyboard.delay(2000);
  for(;;){ /*empty*/ }
  }
  else if(c=='0')
  {
  digitalWrite(0, LOW);
  digitalWrite(1, LOW);
  }
 }
 delay(500);
}`

RyanLoringCooper, 2021-09-14:
It looks like this library and the DigisparkKeyboard both define an interrupt service routine for the same pin, See the ISR defined for PCINT0_vect from [twi.cpp:821](https://github.com/lucullusTheOnly/TinyWire/blob/master/twi.cpp) and USB_INTR_VECTOR from [usbdrvasm165.inc:39](https://github.com/digistump/DigisparkArduinoIntegration/blob/master/libraries/DigisparkKeyboard/usbdrvasm165.inc). The compiler must be translating PCINT0_vect and USB_INTR_VECTOR into the same value (__vector_2). Additionally, you can see where USB_INTR_VECTOR is defined in [usbconfig.h:377](https://github.com/digistump/DigisparkArduinoIntegration/blob/8d0b0de6d208c55c77d88c09409d619309e85201/libraries/DigisparkKeyboard/usbconfig.h), which is PCINT0_vect.

To me it looks like the libraries are incompatible with each other for this reason.

Giova02, 2021-09-14:
@RyanLoringCooper Do you know if there could be any solution to this issue?

Giova02, 2021-09-14:
> It looks like this library and the DigisparkKeyboard both define an interrupt service routine for the same pin, See the ISR defined for PCINT0_vect from [...]

I found out that in the usbconfig.h it's written: 

` #if defined (__AVR_ATtiny45__) || defined (__AVR_ATtiny85__) 
#define USB_INTR_CFG            PCMSK
#define USB_INTR_CFG_SET        (1<<USB_CFG_DPLUS_BIT)
#define USB_INTR_ENABLE_BIT     PCIE
#define USB_INTR_PENDING_BIT    PCIF
#define USB_INTR_VECTOR         SIG_PIN_CHANGE
#endif

#if defined (__AVR_ATtiny87__) || defined (__AVR_ATtiny167__)
#define USB_INTR_CFG            PCMSK1
#define USB_INTR_CFG_SET        (1 << USB_CFG_DPLUS_BIT)
#define USB_INTR_CFG_CLR        0
#define USB_INTR_ENABLE         PCICR
#define USB_INTR_ENABLE_BIT     PCIE1
#define USB_INTR_PENDING        PCIFR
#define USB_INTR_PENDING_BIT    PCIF1
#define USB_INTR_VECTOR         PCINT1_vect
#endif`

, so it define the USB_INTR_VECTOR as SIG_PIN_CHANGE and not  PCINT0_vect as you said previously but idk what does it means